PX4/px4_msgs #3: issue in pasring fault_status_bad_Airspeed

xo01, 2020-12-30:
**Related File:**
https://github.com/PX4/px4_msgs/blob/19acde7e8ee2d483ce51643615fcfdee3cf65146/msg/EstimatorStatusFlags.msg#L43

**Current parser output:**
bool fault_status_bad_Airspeed

**Expected parser output:**
bool fault_status_bad_airspeed

**Stack:**
openjdk version "11.0.9.1" 2020-11-04
OpenJDK Runtime Environment (build 11.0.9.1+1-Ubuntu-0ubuntu1.20.04)
OpenJDK 64-Bit Server VM (build 11.0.9.1+1-Ubuntu-0ubuntu1.20.04, mixed mode)
[2.211s] WARNING:colcon.colcon_core.package_selection:ignoring unknown package 'ros1_bridge' in --packages-skip
Starting >>> px4_msgs
--- stderr: px4_msgs                           
CMake Error at /opt/ros/foxy/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:60 (message):
  execute_process(/usr/bin/python3 -m rosidl_adapter --package-name px4_msgs
  --arguments-file
  /ROS/px4_ros_com_ros2/build/px4_msgs/rosidl_adapter__arguments__px4_msgs.json
  --output-dir /ROS/px4_ros_com_ros2/build/px4_msgs/rosidl_adapter/px4_msgs
  --output-file
  /ROS/px4_ros_com_ros2/build/px4_msgs/rosidl_adapter/px4_msgs.idls) returned
  error code 1:

  Error processing 'bool fault_status_bad_Airspeed' of
  'px4_msgs/EstimatorStatusFlags': ''fault_status_bad_Airspeed' is an invalid
  field name.  It should have the pattern
  '^(?!.*__)(?!.*_$)[a-z][a-z0-9_]*$''

  Traceback (most recent call last):

    File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
      exec(code, run_globals)
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/main.py", line 53, in main
      abs_idl_file = convert_to_idl(
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/__init__.py", line 19, in convert_to_idl
      return convert_msg_to_idl(
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/msg/__init__.py", line 28, in convert_msg_to_idl
      msg = parse_message_string(package_name, input_file.stem, content)
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/parser.py", line 512, in parse_message_string
      fields.append(Field(
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/parser.py", line 353, in __init__
      raise NameError(

  NameError: 'fault_status_bad_Airspeed' is an invalid field name.  It should
  have the pattern '^(?!.*__)(?!.*_$)[a-z][a-z0-9_]*$'

Call Stack (most recent call first):
  /opt/ros/foxy/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
  CMakeLists.txt:44 (rosidl_generate_interfaces)

---
Failed   <<< px4_msgs [23.1s, exited with code 1]

Summary: 0 packages finished [24.7s]
  1 package failed: px4_msgs
  1 package had stderr output: px4_msgs
  1 package not processed

TSC21, 2020-12-30:
Interesting. Thanks for reporting. It's weird because the field on the uORB topic is correct - https://github.com/PX4/PX4-Autopilot/blob/master/msg/estimator_status_flags.msg#L43. So this might be something in the uORB to IDL converter. I will take a look. In the mean time if it is blocking you, just change the IDL field manually to snake_case (lower case in all letters).

xo01, 2020-12-30:
@TSC21 Thank you for your help. and yes it seems the issue with the uORB to ROS msg converter.

TSC21, 2021-01-12:
Solved in https://github.com/PX4/px4_msgs/commit/58b83feb39eef72d030ee361a2b6df12144e2b88.

closed 2021-01-12 by TSC21