PX4/px4_msgs #7: Trying to add a custom made topic to px4_msgs

GuillermoHerreraF, 2021-06-11:
Hello @TSC21. I hope you are doing fine. 

I'm trying to add a custom topic to `px4_msgs` based on a uORB topic that I created in the PX4 firmware. However, I have not yet managed to make it work. I know the uORB topic on the PX4 firmware works fine because I manage to use it with MAVLink. However, I would like to add some autonomous control capability with ROS2 running on a companion computer. 
The client is running on physical hardware (CUAV V5 nano) and I'm running the agent on a Linux machine (Ubuntu 20 and ROS Foxy). I'm running the bridge in serial configuration (UART). 

First of all I generated the `CustomCmd.msg` with the `uorb_to_ros_msgs.py` script located in `px4-ws/msg/tools`. I copied the generated file into the `px4_ros_com_ros2/src/px4_msgs/msg` folder. 

I also went to the generated agent folder in the PX4 firmware workspace, `px4-ws/build/<target-platform>/src/modules/micrortps_bridge/micrortps_agent` and copied the 6 generated files into the `px4_ros_com_ros2/src/px4_ros_com/src/micrortps_agent` folder

- `custom_cmd.cpp`
- `custom_cmd.h`
- `custom_cmdPubSubTypes.cpp`
- `custom_cmdPubSubTypes.h`
- `custom_cmd_Subscriber.cpp`
- `custom_cmd_Subscriber.h`

I copied the generated `custom_cmd.idl` and `CustomCmd.msg` files into the `px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/msg` folder.

I copied the generated `custom_cmd.idl` into the `px4_ros_com_ros2/buildl/px4_msgs/rosidl_adapter/px4_msgs/msg` folder and added the following line of code, `/home/user/px4_ros_com_ros2/build/px4_msgs/rosidl_adapter/px4_msgs:msg/crane_cmd.idl`, into the `px4_ros_com_ros2/buildl/px4_msgs/rosidl_adapter/px4_msgs.idls` file.

On the PX4 firmware side, I added the following lines into the `uorb_rtps_message_ids.yaml`

```
  - msg: custom_cmd
    id: 141
    receive: true
```

And on the agent side I added 

```
  - id: 141
    msg: CustomCmd
    receive: true
```

I recompiled the `px4_ros_com_ros2` workspace after all these changes but I still do not manage to see my custom topic on the list which is displayed when running the agent. 

I would appreciate any help as I've been stuck with this issue for a few days now.
Thank you in advance!

TSC21, 2021-06-11:
> First of all I generated the CustomCmd.msg with the uorb_to_ros_msgs.py script located in px4-ws/msg/tools. I copied the generated file into the px4_ros_com_ros2/src/px4_msgs/msg folder.

So far so good.

> I also went to the generated agent folder in the PX4 firmware workspace, `px4-ws/build/<target-platform>/src/modules/micrortps_bridge/micrortps_agent` and copied the 6 generated files into the [...]

This is not how this is done. After you copied `.msg` to `px4_msgs`, you have to rebuild `px4_msgs`, and it will automatically generate the idl files and typesupport used by `px4_ros_com` to generate the micrortps_agent code.

> On the PX4 firmware side, I added the following lines into the `uorb_rtps_message_ids.yaml`
> 
> ```
>   - msg: custom_cmd
>     id: 141
>     receive: true
> ```
> 
> And on the agent side I added
> 
> ```
>   - id: 141
>     msg: CustomCmd
>     receive: true
> ```

This step is also correct. So all together:
1. Create the uORB `,msg` file under `PX4-Autopilot/msg/`
1. Generate the ROS `.msg` using the `uorb_to_ros_msgs.py` script and copy it to `px4_msgs` under you colcon/ament workspace
1. Do the changes on both `yaml` files in `PX4-Autopilot/msg/tools` and `px4_ros_com/templates` as you did above
1. Build your colcon/ament workspace with both updated `px4_ros_com` and `px4_msgs`

GuillermoHerreraF, 2021-06-12:
Thanks a lot for your help @TSC21 and your fast response. I managed to add the topic to the list when I run the agent. I will verify now that my messages actually arrive to the board.

TSC21, 2021-06-14:
> Thanks a lot for your help @TSC21 and your fast response. I managed to add the topic to the list when I run the agent. I will verify now that my messages actually arrive to the board.

Ok. If all is good, please feel free to close this issue.

karishmapatnaik, 2021-06-18:
Thank you for the response! I could get it working too by following the above steps.

karishmapatnaik, 2021-06-21:
Hi, 
Continuing on this issue, I tried the above and I can see the topics in the Rqt window under topics of the agent. However, as soon as I start the client node from px4 console, the following appears on the agent terminal and there is no data inflow when subscribed to this new topic:
Unexpected topic ID 'xxx' to publish Please make sure the agent is capable of parsing the message associated to the topic ID 'xxx'. Is this a warning of some sort or is it directly related to the issue I am facing? I built the colcon workspace multiple times and the yaml files have been changed to include the msg IDs on both px4/msg/tools and px4_ros_com/templates . Also, I could verify from an external module that the new uORB topic is successfully publishing data, but I have issue sending it over to ROS2.  Any help is appreciated, thanks a lot. 
The msg IDs are, on client side:
  - msg: custom_msg
    id: 165
    send: true 
On agent side
  - id: 165
    msg: CustomMsg
    send: true

karishmapatnaik, 2021-06-21:
Update: I had to clean and build the colcon ws and it works!

GuillermoHerreraF, 2021-06-30:
Hello again @TSC21 
Hope you are doing well.
I'm trying to publish (directly from the terminal) to the custom cmd topic I created a few weeks ago. When I start publishing, in the terminal running the agent it indicates "subscriber matched", but I'm not receiving the commands on the PX4 firmware side. 
Any clue what may be happening here?

TSC21, 2021-06-30:
> Hello again @TSC21
> Hope you are doing well.
> I'm trying to publish (directly from the terminal) to the custom cmd topic I created a few weeks ago. When I start publishing, in the terminal running the agent it indicates "subscriber matched", but I'm not receiving the commands on the PX4 firmware side.
> `[micrortps_agent] CustomCmd subscriber matched`
> Any clue what may be happening here?

Hard to tell without more context. Do you have the `micrortps_client` running and setup properly? Do you have that same topic being published on the `micrortps_client`? You need to make sure that your client is able to receive the message ID 141.

Also this is not related to this repo in specific so I ask you to move this to `px4_ros_com` issues.

closed 2021-06-30 by TSC21