PX4/px4_msgs #9: Issue in CMakeFlies.txt

lymnxn, 2022-01-25:
I face some problem when I complie it. My cmake version is 3.16.3.
Starting >>> px4_msgs
--- stderr: px4_msgs                         
CMake Error at CMakeLists.txt:25 (if):
  if given arguments:

    "MATCHES" "ardent" "OR" "MATCHES" "bouncy" "OR" "MATCHES" "crystal"

  Unknown arguments specified

---
I change line 25 and line 52 to :

```
if((${ROS_DISTRO} MATCHES "ardent") OR (${ROS_DISTRO} MATCHES "bouncy") OR (${ROS_DISTRO} MATCHES "crystal"))
```
and solve it

sumedhreddy90, 2022-07-23:
Bravo @lymnxn  `if((${ROS_DISTRO} MATCHES "ardent") OR (${ROS_DISTRO} MATCHES "bouncy") OR (${ROS_DISTRO} MATCHES "crystal"))`

This helped me to solve the issue. Thanks mate!

sumedhreddy90, 2022-07-23:
Raised a PR https://github.com/PX4/px4_msgs/pull/12