PX4/px4_msgs #17: Visual Odometry

321thijs123, 2022-10-28:
Previously I used timesync and vehicle_visual_odometry to take fly using external position estimates from a motion capture system. These topics have been removed and documentation on it has not been updated.

How can visual odometry be used now?

dagar, 2022-10-28:
The topic is `/fmu/in/vehicle_visual_odometry` and the type is https://github.com/PX4/px4_msgs/blob/main/msg/VehicleOdometry.msg.

https://github.com/PX4/PX4-Autopilot/blob/498937c56cd1b1a676bb93fa22ee7b7bc99c69cc/src/modules/microdds_client/dds_topics.yaml#L76-L77

Things are still a bit in flux with the switch to micro XRCE DDS, but please let me know how it goes. Separately for PX4 itself we've got a number of changes slowly coming in improving EKF2 VIO. https://github.com/PX4/PX4-Autopilot/pull/19128

dagar, 2022-10-28:
For the timestamp (and timestamp_sample) just use `get_clock()->now()` converted to microseconds. The XRCE agent (micro-ros-agent) handles the actual time sync with microdds_client in PX4.

https://github.com/PX4/px4_ros_com/blob/abc59d8381f7bc944af4c73b6c58dcc85ac5ef45/src/examples/offboard/offboard_control.cpp#L180

321thijs123, 2022-10-28:
Thanks! This answers my question

closed 2022-10-28 by 321thijs123