PX4/px4_msgs #24: Release/1.13 build failing

davidemorazzo, 2023-03-19:
I am trying to build the code in the `release/1.13` branch and it is failing. The output error from colcon is 
```
--- stderr: px4_msgs                             
In file included from /home/judocero/Scrivania/uav_config/gz_groundtruth/build/px4_msgs/rosidl_generator_c/px4_msgs/msg/detail/vehicle_command__functions.h:19,
                 from /home/judocero/Scrivania/uav_config/gz_groundtruth/build/px4_msgs/rosidl_generator_c/px4_msgs/msg/detail/vehicle_command__functions.c:4:
/home/judocero/Scrivania/uav_config/gz_groundtruth/build/px4_msgs/rosidl_generator_c/px4_msgs/msg/detail/vehicle_command__struct.h:365:3: error: expected identifier or ‘(’ before ‘/’ token
  365 |  */
      |   ^
gmake[2]: *** [CMakeFiles/px4_msgs__rosidl_generator_c.dir/build.make:6077: CMakeFiles/px4_msgs__rosidl_generator_c.dir/rosidl_generator_c/px4_msgs/msg/detail/vehicle_command__functions.c.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:213: CMakeFiles/px4_msgs__rosidl_generator_c.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< px4_msgs [6.48s, exited with code 2]
```
I fixed this by removing the extra `*/` that was giving the error. There are a couple present that make the build fail

beniaminopozzan, 2023-03-19:
You are trying to build `release/1.13` on ROS 2 Humble or Rolling, right?
PX4 version `1.13` (and therefore px4_msgs and px4_ros_com) supports up to ROS 2 Foxy

davidemorazzo, 2023-03-20:
Yes I am trying with ROS2 Humble. How can I use then RTPS on Humble?

beniaminopozzan, 2023-03-21:
To use v1.13 on Humble you need to

- fix the `*/` that you spotted
- see if `px4_ros_com` works. From my experience it doesn't. But if you manage to have a working solution you are more than welcome to raise a PR!

closed 2023-03-30 by davidemorazzo