PX4/px4_msgs #25: Update comments in VehicleCommands

audrow, 2023-03-31:
It seems that the comments in https://github.com/PX4/px4_msgs/blob/main/msg/VehicleCommand.msg have fallen out of sync with the messages being used. Maybe the comments could be updated, or if not deleted (especially the incorrect information).

For example, the comments with `VEHICLE_CMD_DO_SET_MODE` say that it takes only one argument,
https://github.com/PX4/px4_msgs/blob/main/msg/VehicleCommand.msg#L32

But the command on Mavlink takes up to three parameters.

https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LAND

Also, some comments list the inputs for the command. In the case that no inputs are listed, I assume that the command takes no parameters, but this is not true in every case. For example, 

https://github.com/PX4/px4_msgs/blob/main/msg/VehicleCommand.msg#L32

Has two arguments documented here:

https://mavlink.io/en/messages/common.html#MAV_CMD_DO_VTOL_TRANSITION

audrow, 2023-03-31:
As a side note, it is not clear how the https://github.com/PX4/px4_msgs/tree/ros2 branch is different from the main branch. They both seem to be ROS 2, but the ros2 branch doesn't seem to have been updated since 2019.

kjrusscher, 2023-04-06:
Thanks for pointing out that the _ros2_ branch is a stale branch; that saved me a lot of time. The file generated form https://github.com/PX4/px4_msgs/blob/main/msg/VehicleCommand.msg contain some errors that give compile errors. The _main_ branch does not give any compile errors. So I guess that we have to use the _main_ branch for ROS2.

beniaminopozzan, 2023-05-09:
@TSC21 , can we safely remove the `ros2` branch?

Thanks @audrow !
BTW, all the `.msg` files in this repo are directly copied form the ones in the main firmware, there is a 1 : 1 correspondence, so this issue should be opened in https://github.com/PX4/PX4-Autopilot