PX4/px4_msgs #41: Release binaries of px4_msgs due to long compilation time

Ryanf55, 2024-08-01:
# Request 

Please consider releasing tagged binaries for px4_msgs.
The CI time for this package on the last job was over 5 hours.
Github's [limit for their runners is 6 hours](https://docs.github.com/en/actions/administering-github-actions/usage-limits-billing-and-administration#usage-limits)
Because binaries are not available, I must build this from source, and it takes a long time.

# Problems encountered

Because binaries are not available, I must build this from source, and it takes a long time.
I'm limited to 2 hours per job on the CI solution I have.

# Proposed Solutions

* Release binaries to ROS 2 humble using [these instructions](https://docs.ros.org/en/humble/How-To-Guides/Releasing/First-Time-Release.htmll)
* Add ccache to the Gihtub environment (only useful for those forking this in Github)

# Complexities

Because PX4 does not promise a message-stable ABI within an LTS ROS distro, I am not sure whether the ROS build farm will allow you to support multiple versions, or how this will work with rosdep's lack of versioning support. Maybe this will require PX4 to host a PPA and you can do `sudo apt install ros-humble-px4-msgs-f31e301b3`, where that corresponds to this build hash; https://github.com/PX4/px4_msgs/commit/f31e301b34272336c9f26fbd33023814d7a1b4ba

afxalz, 2024-08-01:
I am having a lot trouble building the package. The compile time along with the debug time is quite large. Releasing the binaries would be extremely useful to accelerate the development process. I think the community will greatly appreciate this help.

dagar, 2024-08-01:
I'm not opposed to packaging if we do it corresponding to PX4 Autopilot stable releases (v1.14.x, etc). What would be the right way to package/release px4_msgs corresponding to different versions of PX4 simultaneously? Would they each need to be a distinct interface package with version naming or could we do it such that they all provide px4_msgs, but they can't be installed simultaneously. 

We could also easily limit px4_msgs to the small subset people are even using externally. For example here's what we currently have by default for microXRCE-DDS. https://github.com/PX4/PX4-Autopilot/blob/75ce550db39cc5d61ba970ca5f4b6096a287827e/src/modules/uxrce_dds_client/dds_topics.yaml

The potential downside is that I want it to be super easy for developers to privately add whatever random message they might want, but I think we could start with a reasonable guess and still cut it down the absurd build time by like 80-90%.

@Jaeyoung-Lim @beniaminopozzan @TSC21

dagar, 2024-08-01:
> Because binaries are not available, I must build this from source, and it takes a long time.
I'm limited to 2 hours per job on the CI solution I have.

@Ryanf55 separately from my comment above, have you tried building with ninja instead of make? 

``` Bash
$ colcon build --cmake-args " -GNinja"
```

Ryanf55, 2024-08-01:
> I'm not opposed to packaging if we do it corresponding to PX4 Autopilot stable releases (v1.14.x, etc). What would be the right way to package/release px4_msgs corresponding to different versions of [...]

I don't have an easy answer on how to break ABI during a distro. It's not part of the standard ROS way of doing it, so I recommend we talk to some of the infrastructure maintainers at ROS on how to accomplish that in a way that they would accept on their build farm.

The way that ROS handles when you have different message definitions is extremely poor (silently accepting junk data, or runtime crashes), so we have to make sure it's super obvious to users how to keep PX4 and your messages compatible.

I'll report back in Ninja can speed it up, I was assuming we already had that enabled everywhere internally.

Jaeyoung-Lim, 2024-08-01:
@Ryanf55 Just be aware that px4_msgs are not supposed to be stable, and the expectation is that every px4 commit will break the compatibility of px4_msgs

Ryanf55, 2024-08-01:
Absolutely, I get that PX4 wants to change messages. Totally fine. 

Here's a proposed workflow:

I have a repos file:

```yaml
repositories:
   px4_msgs:
      type: git
      url: git@aimlgit.aerovironment.com:AVACore/publicexternals/px4_msgs.git
      version: v1.14.x
   PX4-Autopilot:
      type: git
      url: git@aimlgit.aerovironment.com:AVACore/publicexternals/PX4-Autopilot.git
      version: v1.14.x
```

When I 
`rosdep install --from-paths src`
Then, I expect the following to occur:
`apt install ros-humble-px4-msgs-v1.14`
Tomorrow, `ros-humble-px4-msgs-v1.15` could be released that breaks ABI.

Although with a PPA you could do `apt install ros-humble-px4-msgs~1.14`, rosdep doesn't support versioning to apt.

Then, in CI, when I run colcon build, I can do a `colcon build --packages-skip px4_msgs_v1.14` because it's available through apt.

Headers could be installed as `px4_msgs/msg/MyMsg.hpp` to preserve existing header paths.

We would need to decide whether you do `find_package(px4_msgs CONFIG 1.14)` or `find_package(px4_msgs_v1.14)`

Ryanf55, 2024-08-02:
As far as taking the route of limiting which messages are generated, what about adding CMake options for PX4_MIN_MSGS and PX4_ALL_MSGS. 

PX4_ALL_MSGS is the list of all interfaces, while MIN is only the ones used by XRCE DDS. 

In order for a user to choose which messages, they set a variable PX4_GENERATED_MSGS to either MIN or ALL, and they can also override/remove/append items to those lists if they want to remove or add messages. 

Then default build could be all messages, but now we have a one-liner through --cmake-args to only generate the interfaces used by ROS and cut the build time down.

dagar, 2024-08-02:
If you're suggesting adding that here (PX4/px4_msgs) that sounds like a harmless workaround for now.
I was talking about only extracting the minimum (or reasonable) subset from PX4-Autopilot in the first place.

Ryanf55, 2024-08-07:
It looks like `colcon` already enables building with all cores and a parallel build, regardless of which generator you use.
https://robotics.stackexchange.com/questions/97896/colcon-build-number-of-threads

I checked why the CI is taking 5 hours here, and did not see any obvious problems in the Github actions used.

beniaminopozzan, 2024-10-06:
While it is true that binaries for this package would be great, the 6h build time on the CI were caused by `ros-tooling/action-ros-ci` running tests after compiling.
I did not investigate where exacly the issue is, but adding `skip-test` (https://github.com/ros-tooling/action-ros-ci?tab=readme-ov-file#skip-tests) reduced the CI time to 11 minutes.
PR for the fix is https://github.com/PX4/px4_msgs/pull/34

asherikov, 2024-12-24:
@beniaminopozzan it would be better to drop ADD_LINTER_TESTS in https://github.com/PX4/px4_msgs/blob/main/CMakeLists.txt#L32C2-L32C18 instead, it is a more general workaround, not specific to github. To my understanding those are more about testing the code generator rather than the messages.

mcamurri, 2026-08-19:
Is there any news on this? 

Given that the only option now is to compile `px4_msgs` from source, and provided that `px4_msgs` is just ... messages and services, why don't you simply choose and release *a version* of that package to the ROS buildfarm *without* the PX4-autopilot? 

Then it is up to the user to choose whether to use the APT packaged version or compile and customize the version they want and clone/build the matching version of PX4-Autopilot (which they have to do anyway).

This way at least the users who just need the messages for their custom application (like myself) have something to work with without having to build `px4_msg` from source every time just because their code needs the messages.

Please see also my request here: https://discourse.openrobotics.org/t/missing-px4-msgs-from-apt/57491

BTW, IMHO this packages shouldn't be open to users to add their own messages ... they can always make another custom package for that. Nobody would think of adding more messages to, say, `geometry_msgs`, they would just make their own separate package.

Ryanf55, 2026-08-19:
> Is there any news on this?
>
> Given that the only option now is to compile `px4_msgs` from source, and provided that `px4_msgs` is just ... messages and services, why don't you simply choose and [...]

Agree with all your points.

Jaeyoung-Lim, 2026-08-19:
@mcamurri Thanks for the bump on the topic. @Ryanf55 what are the blockers on distributing the binaries? I think at least of the stable releases it would make sense we have it released on the build farms.

@mrpollo FYI

TSC21, 2026-08-19:
This is already in place missing only a couple of things: https://github.com/PX4/px4_msgs/pull/68

Ryanf55, 2026-08-20:
> [@mcamurri](https://github.com/mcamurri) Thanks for the bump on the topic. [@Ryanf55](https://github.com/Ryanf55) what are the blockers on distributing the binaries? I think at least of the stable [...]

Agree, but once concern is how you handle PX4 versions not in sync with ROS versions. IE, in ROS 2 Jazzy, you might have 5 different PX4 releases with no message ABI stability between them.

Is there a plan for how you will break ABI in PX4, and maintain it within the ROS release?

mrpollo, 2026-08-20:
Can we build our own binaries on our infra instead of using the ros build farm?

Ryanf55, 2026-08-20:
> Can we build our own binaries on our infra instead of using the ros build farm?

Yes, ROS allows that, but if you want to do that to solve the binary version release cadence issue, I don't see that helping.

mrpollo, 2026-08-20:
That's not the reason, I want to own the infra and delivery, regardless of what we bundle into a ROS distro

asherikov, 2026-08-24:
A few sidenotes:
- PX4 sitl can also be built and distributed as a binary package, it should be a low effort addition to  px4_msgs packages.
- Regarding the infra ownership: ROS uses `/opt/ros/<ros_distro>` paths, with a dedicated build farm it should be possible to do something like `/opt/px4/v1.16/`. The buildfarm setup is rather involved process though.

mcamurri, 2026-09-01:
A few suggestions:
- `px4_msgs` is primarily ROS messages, therefore it would make sense to have it in the ROS buildfarm and/or your infrastructure as a mirror (if you want to own it), but the package should be searchable and obtainable from the ROS ppa IMHO. 
- I would start with the `msgs` only because they don't depend on anything else. It would already help to be able to get them through `apt install ros-jazzy-px4-msgs` instead of manually building them
- I would discourage versioning in the path or name of the package, there will be a version already attached to the APT package signature and the `package.xml` manifest, so it is enough for the user to tell which version it is. Just pick the one you think is most suitable for `Jazzy`
- If a user needs a different version, they can just avoid using APT and do the manual cloning, as it is now already. 
- Once the above works, I would take care of adding the other packages matching the same version

As noted by the reply I got on discourse, the package is already in the buildfarm, but it is missing the release target, so you get only source and docs, but not the binaries:
https://github.com/ros/rosdistro/blob/3652e6c70f3d8bcccb0e00d731ef18a5243d7f90/jazzy/distribution.yaml#L8251-L8261

asherikov, 2026-09-01:
> I would discourage versioning in the path or name of the package, there will be a version already attached to the APT package signature and the package.xml manifest, so it is enough for the user to tell which version it is. Just pick the one you think is most suitable for Jazzy

Only one version of prebuilt PX4 messages for ROS distro would greatly reduce the value of binary packages  -- there is no point in maintaining different workflows for various PX4 versions (this is not uncommon in practice). It is perfectly ok to have packages with version numbers in their names, e.g., gcc.

Jaeyoung-Lim, 2026-09-01:
> Only one version of prebuilt PX4 messages for ROS distro would greatly reduce the value of binary packages -- there is no point in maintaining different workflows for various PX4 versions (this is not uncommon in practice).

@asherikov Given that PX4 run with hardware, it will be challenging to associate px4 versions with ROS 2 versions. Therefore, we would probably need to publish binaries of multiple versions per ROS distro in order for people to use it