PX4/px4_msgs #42: Colcon build failed:Starting >>> px4_msgs  --- stderr: px4_msgs

genghaijian001, 2024-08-04:
I've never encountered this problem before.
And I don't know how to solve it.
Can anyone pls help me,thank you
:)
geng@geng-VirtualBox:~/ws_sensor_combined$ colcon build

Starting >>> px4_msgs

--- stderr: px4_msgs                               

Traceback (most recent call last):

  File "/home/geng/ws_sensor_combined/build/px4_msgs/ament_cmake_python/px4_msgs/setup.py", line 4, in <module>

    setup(

  File "/home/geng/.local/lib/python3.10/site-packages/setuptools/__init__.py", line 108, in setup

    return distutils.core.setup(**attrs)

  File "/home/geng/.local/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 184, in setup

    return run_commands(dist)

  File "/home/geng/.local/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 200, in run_commands

    dist.run_commands()

  File "/home/geng/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 970, in run_commands

    self.run_command(cmd)

  File "/home/geng/.local/lib/python3.10/site-packages/setuptools/dist.py", line 945, in run_command

    super().run_command(command)

  File "/home/geng/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 989, in run_command

    cmd_obj.run()

  File "/home/geng/.local/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 310, in run

    self.find_sources()

  File "/home/geng/.local/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 318, in find_sources

    mm.run()

  File "/home/geng/.local/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 544, in run

    self.prune_file_list()

  File "/home/geng/.local/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 610, in prune_file_list

    base_dir = self.distribution.get_fullname()

  File "/home/geng/.local/lib/python3.10/site-packages/setuptools/_core_metadata.py", line 266, in get_fullname

    return _distribution_fullname(self.get_name(), self.get_version())

  File "/home/geng/.local/lib/python3.10/site-packages/setuptools/_core_metadata.py", line 284, in _distribution_fullname

    canonicalize_version(version, strip_trailing_zero=False),

TypeError: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'

gmake[2]: *** [CMakeFiles/ament_cmake_python_build_px4_msgs_egg.dir/build.make:70: CMakeFiles/ament_cmake_python_build_px4_msgs_egg] Error 1

gmake[1]: *** [CMakeFiles/Makefile2:447: CMakeFiles/ament_cmake_python_build_px4_msgs_egg.dir/all] Error 2

gmake: *** [Makefile:146: all] Error 2

---

Failed   <<< px4_msgs [11.3s, exited with code 2]

Summary: 0 packages finished [11.6s]

  1 package failed: px4_msgs

  1 package had stderr output: px4_msgs

  1 package not processed

ba1Sta, 2024-08-07:
I have encountered the same issue. First please try to check your current version of packaging
```
pip list | grep packaging
```
if the version of packaging is <22.0, please upgrade it to 22.0
```
pip install packaging==22.0
```
Then, remove log, build and install folders in the workspace and `colcon build` again. Hope this can help you to fix this issue too.

Kametor, 2024-08-28:
@ba1Sta I got same error, your solution is solved my problem too, thanks.