PX4/px4_msgs #44: colon build failed

JT-Klenke, 2024-09-28:
Starting >>> px4_msgs
[Processing: px4_msgs]                             
--- stderr: px4_msgs                               
In file included from /home/jt/ws_sensor_combined/build/px4_msgs/rosidl_generator_c/px4_msgs/msg/detail/vehicle_command__functions.h:19,
                 from /home/jt/ws_sensor_combined/build/px4_msgs/rosidl_generator_c/px4_msgs/msg/detail/vehicle_command__functions.c:4:
/home/jt/ws_sensor_combined/build/px4_msgs/rosidl_generator_c/px4_msgs/msg/detail/vehicle_command__struct.h:719:3: error: expected identifier or ‘(’ before ‘/’ token
  719 |  */
      |   ^
gmake[2]: *** [CMakeFiles/px4_msgs__rosidl_generator_c.dir/build.make:6553: CMakeFiles/px4_msgs__rosidl_generator_c.dir/rosidl_generator_c/px4_msgs/msg/detail/vehicle_command__functions.c.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:213: CMakeFiles/px4_msgs__rosidl_generator_c.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< px4_msgs [46.2s, exited with code 2]
                                 
Summary: 0 packages finished [46.3s]
  1 package failed: px4_msgs
  1 package had stderr output: px4_msgs
  1 package not processed

When following [these instructions](https://docs.px4.io/main/en/ros2/user_guide.html#build-ros-2-workspace) I get the above error. I've tried redoing all previous install instructions and I've made sure to source my ROS (harmonic) install. I'm on Ubuntu 22.04

JT-Klenke, 2024-09-28:
The issue was the */ at the end of line 85 on VehicleCommand.msg

beniaminopozzan, 2024-10-06:
Hi @JT-Klenke , the issue has been fixed in in `main` and also `release/1.15`.

closed 2024-10-06 by beniaminopozzan