PX4/px4_msgs #53: [RTPS_READER_HISTORY Error] Gazebo & Error When Running offboard_control.py

DrChanChan, 2025-03-20:
Hi, thank you so much for sharing your code. I really appreciate your work!

However, I'm encountering an issue when running:

ros2 run px4_ros_com offboard_control.py

I get the following error:

[image]

The drone just spins its propellers in place but does not take off in Gazebo.
I’ve already updated the version of px4_msgs to 1.15 to match the PX4 version I’m using, but the issue persists.
For reference, I’m using ROS2 Humble.
Any insights or suggestions would be greatly appreciated. Thanks in advance!

beniaminopozzan, 2025-03-20:
Hi @DrChanChan , 

> I’ve already updated the version of px4_msgs to 1.15 to match the PX4 version I’m using, but the issue persists

Have you just checked out the branch `release/1.15` or you manually copied the messages definition from PX4 v1.15 to the px4_msgs repo? I'm asking this because `release/1.15` is slightly outdated and this may cause the issue.

Could you try again using https://github.com/PX4/px4_msgs/pull/54 please?

DrChanChan, 2025-03-20:
[image]

Thank you, the drone is up!

As you can see in the picture, there is still an [RTPS_READER_HISTORY Error]((88, 87) --> (220, 207)), but the drone itself seems to be moving, so that shouldn't be a problem. 

If you still have any advice, please let me know! If you run into any problems along the way, do you mind if I post about it?

beniaminopozzan, 2025-03-23:
What version of ROS and MicroDDS Agent are you running and how did you compile the Agent?

DrChanChan, 2025-03-23:
Overall, I followed this PX4_1.15 guide : ("**https://docs.px4.io/v1.15/en/ros2/user_guide.html#setup-micro-xrce-dds-agent-client**")

0. PX4_version : **1.15.4** ("git clone https://github.com/PX4/PX4-Autopilot.git --recursive")
1. ROS2 version: **ROS2_humble**
2. MicroDDS Agent version: **2.4.2** ("git clone -b 2.4.2 https://github.com/eProsima/Micro-XRCE-DDS-Agent.git")
3. compile agent: (If I understand correctly, are you referring to how to spawn the drone or the DDS Agent command?)

Drone : make px4_sitl gz_x500
Activate DDS Agent : MicroXRCEAgent udp4 -p 8888