PX4/px4_msgs #55: Irregular message stream with px4 release 1.15.4 and px4_msgs release 1.15

IssoufGit, 2025-04-08:
Hi,
I had some issues with reading out ros2 topic data with the following setup:

PX4 version: release v1.15.4 (px4_fmu-v5-default)
FCU: Pixhawk 4
Companion computer OS: Ubuntu 22.04
ROS2 version: Humble
Serial link: FT231X, connected via TELEM2

The message stream worked fine when using PX4 v1.14.4 and px4_msgs release 1.14.
However, when I updated the px4 version using QgroundControl to v1.15.4, and updated the px4_msgs to release 1.15:
1. No fmu/out/* topic was listed after _ros2 topic list_. (Only fmu/in/... topics were listed.) when time sync was enabled.
2. When **time sync was disabled**, the fmu/out topics were listed after the _ros2 topic list_. However, when ros2 topic echo was called on topic /fmu/out/vehicle_attitude, the stream was not regular (about 1~2Hz regular disruption). Changing the baudrate from 921600 to 3000000 had no impact. The data rate of fmu/out/vehicle_attitude was lower than 47Hz. 

**Note**: I read a lot about message compatibility between the versions of px4 and px4_msgs. By checking out the correct px4_msgs branch, I am sure I was using the correct versions.
I went back to px4 v1.14.4, which is working fine. Does anyone have suggestions on how to make px4 v1.15.4 work with px4_msgs release/1.15 on real hardware (no SITL)?

dbaldwin, 2025-05-05:
I'll share what we did to overcome this for testing purposes. Right now we're using Pi5 companion computer on our DEXI drone with PX4 1.15.4 and ran into the same issue as you did. We cloned PX4 firmware:

```
git clone https://github.com/PX4/PX4-Autopilot.git
cd PX4-Autopilot
git checkout v1.15.4
```

Deleted the existing messages and service files in our px4_msgs package that was cloned into our workspace:

```
rm -rf ~/dexi_ws/src/px4_msgs/msg/*
rm -rf ~/dexi_ws/src/px4_msgs/srv/*
```

Then copied the msg/* and srv/* files from PX4 autopilot into our px4_msgs folder in our ROS2 workspace:

```
cp msg/* ~/dexi_ws/src/px4_msgs/msg/
cp srv/* ~/dexi_ws/src/px4_msgs/srv/
```

Then:

```
cd ~/dexi_ws
colcon build --packages-select px4_msgs
```

Finally:

```
root@raspberrypi:/home/dexi/dexi_ws/PX4-Autopilot# ros2 topic hz /fmu/out/vehicle_attitude
WARNING: topic [/fmu/out/vehicle_attitude] does not appear to be published yet
average rate: 100.081
	min: 0.005s max: 0.017s std dev: 0.00177s window: 102
average rate: 100.025
	min: 0.005s max: 0.017s std dev: 0.00170s window: 203
average rate: 100.027
	min: 0.005s max: 0.017s std dev: 0.00166s window: 304
average rate: 100.016
	min: 0.005s max: 0.017s std dev: 0.00166s window: 405
average rate: 100.008
	min: 0.005s max: 0.017s std dev: 0.00170s window: 506
average rate: 100.044
	min: 0.005s max: 0.017s std dev: 0.00171s window: 607
```

Not the cleanest solution but it worked for us. Good luck!

IssoufGit, 2025-05-09:
Thank you very much for sharing your solution. I will try it out.

beniaminopozzan, 2025-07-19:
v1.15 message definitions have been updated by #54

closed 2025-07-19 by beniaminopozzan